danrubel/rpi_gpio.dart #1: deferred type 'rpi.RpiHardware' is not yet loaded

tomaszkubacki, 2015-07-16:
I've made my project and tried to run [read example](https://github.com/danrubel/rpi_gpio.dart/blob/master/example/read.dart) and got:
```bash
root@raspberrypi:/home/tomek/projects/switchBoard# /opt/dart-sdk/bin/dart main.dart 
Unhandled exception:
'file:///home/tomek/projects/switchBoard/main.dart': malformed type: line 10 pos 25: deferred type 'rpi.RpiHardware' is not yet loaded
    Gpio.hardware = new rpi.RpiHardware();
                        ^
type error.
#0      _TypeError._throwNew (dart:core-patch/errors_patch.dart:58)
#1      main (file:///home/tomek/projects/switchBoard/main.dart:10:25)
#2      _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:261)
#3      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)
```

my project structure seems to be correct and **librpi_gpio_ext.so** has been built
```bash
root@raspberrypi:/home/tomek/projects/switchBoard# tree
.
├── main.dart
├── packages
│   ├── path -> /home/tomek/.pub-cache/hosted/pub.dartlang.org/path-1.3.6/lib
│   └── rpi_gpio -> /home/tomek/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.2.0/lib
├── pubspec.lock
└── pubspec.yaml

3 directories, 3 files
root@raspberrypi:/home/tomek/projects/switchBoard# cd /home/tomek/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.2.0/lib
root@raspberrypi:/home/tomek/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.2.0/lib# ls -la
razem 96
drwxr-xr-x 3 tomek tomek  4096 lip 16 14:07 .
drwx------ 6 tomek tomek  4096 lip 16 12:56 ..
-rwxr-xr-x 1 tomek tomek 59387 lip 16 14:07 librpi_gpio_ext.so
-rw-r--r-- 1 tomek tomek 11647 lip  5 23:35 rpi_gpio.dart
-rw-r--r-- 1 tomek tomek   884 cze 14 14:33 rpi_hardware.dart
-rw-r--r-- 1 tomek tomek  4572 kwi 27 20:04 rpi_pwm.dart
drwxr-xr-x 3 tomek tomek  4096 lip 16 12:56 src
root@raspberrypi:/home/tomek/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.2.0/lib# 
```

danrubel, 2015-07-18:
When I refactored the examples, I forgot to add a wait for the rpi.loadLibrary(). I have published a new [rpi_gpio package](https://pub.dartlang.org/packages/rpi_gpio) (version 0.2.1) that has several fixes including a fixed [read example](https://github.com/danrubel/rpi_gpio.dart/blob/master/example/read.dart).

Please let me know if the revised [read example](https://github.com/danrubel/rpi_gpio.dart/blob/master/example/read.dart) works for you.

tomaszkubacki, 2015-07-19:
yea! it's working perfect  now - thanks a lot :)

closed 2015-07-19 by tomaszkubacki