danrubel/rpi_gpio.dart #12: Overhaul API and update for Dart 2

danrubel, 2018-07-12:
There's a lot of unused C code and unnecessary complexity that I aim to remove in the process of readying this package for Dart 2. The work-in-progress API that I'm working towards has a library of abstract classes ...
```
abstract class Gpio {
  GpioInput input(int physicalPin, [Pull pull = Pull.off]);
  GpioOutput output(int physicalPin);
}

abstract class GpioInput {
  bool get value;
  Stream<bool> get values;
}

abstract class GpioOutput {
  set value(bool newValue);
}
```
... along with a separate library containing a concrete `RpiGpio` implementation of `Gpio` and associated native code to make it all happen.

jimmyff, 2018-08-02:
Really pleased to hear you're updating this package. 

Let me know if you would like some help - although I can't help out with the C code!

danrubel, 2018-08-03:
Much appreciate the offer of help. 
I've partially implemented the above API, but more importantly... is that API enough for your uses?

jimmyff, 2018-08-07:
I've been tracking this package for a while and always been keen to see if I can drive an LED matrix from Dart but not yet given it a go. This is a good excuse to try, so give me a few weeks and I'll get back to you!

danrubel, 2018-08-24:
Just pushed 0.4.0-dev.1 with the new API.
The basic GPIO works, but...

* pulse width modulation does not work yet
* A/D input support not added yet
* only been tested on Raspberry Pi 3

Good progress, but I still have more work to do

danrubel, 2018-09-25:
Just pushed 0.5.0 with new API.

J4ckTh3R1qp3r, 2021-01-10:
Great package! Thank you a lot. Is there any chance to see update with PWM support soon? I am testing this on Pi4b by the way.

danrubel, 2021-01-12:
Glad it's useful to you! I used this package in production but have never needed PWM and thus haven't gotten around to adding it. How do you envision using it? That will give me a better idea of what needs to be added.

And thanks for letting me know that you are on the Pi 4... good to know it works there as well.

J4ckTh3R1qp3r, 2021-01-12:
Actually Dart is my first programming language and Flutter is the first SKD. My needs right now are more about hobby and learning something new and interesting, but not about production or any other serious project/mvp. I want to upgrade a child toy - racing track for two cars. Each car has 2 contacts at the bottom which get 5V and GND (I guess) from contacts on the track and you can control speed of a car with a trigger which is actually a manual resistor. I want to connect track directly to Pi and control the speed of each car with PWM (Angel sdk as API server). I tried to use Dart loops to somehow repeat the PWM process, but they are too slow. If you have any other idea how I can reach my goal please share, that would be great!

danrubel, 2021-01-14:
I did some digging and from it appears that [the 2 PWM pins on RPi 4B](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf) might not [interfere with the audio jack](https://www.raspberrypi.org/forums/viewtopic.php?t=245209) (an issue on earlier Pi models). I don't know for sure and I don't know if you need the audio jack.

I don't think you can drive/power the track directly from the RPi. You'd need some type of relay between the PWM output and the track. I think an optical relay based on the switching frequency needed. It's been quite a while since I played around in this area so consider this advice imperfect at best :-)

It will take me a while but I can look into updating the code to drive LEDs with PWM, but it would be up to you to figure out the rest of the circuit from the PWM to powering the track. Let me know if you are up for that and I can start working towards PWM support in this library.

J4ckTh3R1qp3r, 2021-01-18:
Thank you very much! I already ordered this track and will get it by the end of the week. Will wait for any updates and can help with testing on PRI 4b. This could be the first Dart package that supports PWM 👍🥇🎉

J4ckTh3R1qp3r, 2021-01-25:
Hi! Some updates from me:
1) Each car consumes about 12V 0.6A
2) I connected 12V to a 12V light diod through transistor which was controlled by RPI. 
3) I actually could control the brightness of diod through PWM on python: 
```
import time
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setup(17, GPIO.OUT)

led = GPIO.PWM(17, 500)
led.start(0)

while(1):
 for dc in range(0, 101, 5):
  led.ChangeDutyCycle(dc)
  time.sleep(0.1)

 for dc in range(100, -1, -5):
  led.ChangeDutyCycle(dc)
  time.sleep(0.1)

GPIO.cleanup()
```
4) My schema was wrong as I had only 2-3V as an output from transistor and I started digging this problem.
5) At some point something went wrong as I am total noob and I fried my RPI 😂😜
6) I decided to order another model 4b and a Zero W as It it less expensive for the next BBQ and possibly help to test dart on it.

Well, I think now It should be clear how I will use PWM (frying RPI, yeah! :) ).

danrubel, 2021-01-26:
Ouch! Welcome to hardware 😄 . I've fried my share of chunks of hardware over the years.

I'm opening [a new issue focused on adding PWM support](https://github.com/danrubel/rpi_gpio.dart/issues/18)
so that we can deep drive there without overloading this issue with PWM details.

danrubel, 2021-07-04:
Reworked the code to support PWM in a separate isolate... and published 0.6.0

danrubel, 2022-11-14:
This package has been updated to use `dart:ffi` and now runs on Dart SDK 2.18

danrubel, 2023-10-23:
Just pushed version 0.9.1 with [fixes for gpio inputs on RPi 4](https://github.com/danrubel/rpi_gpio.dart/issues/24)

danrubel, 2024-10-07:
Version 0.10 now relies on `libgpiod` so that it will work on RPi v5.