danrubel/rpi_gpio.dart #16: cant run on armhf

guyluz11, 2019-06-13:
Under armhf architecture cant run the library
`dart example/example.dart `
Error:
`Unhandled exception:
RpiGpio initialization failed: -1
#0      new RpiGpio (package:rpi_gpio/rpi_gpio.dart:26:7)
#1      main (file:///home/pi/newSnap/rpi_gpio.dart/example/example.dart:12:20)
<asynchronous suspension>
#2      _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:300:19)
#3      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:171:12)`
My device is NanoPi NEO2

guyluz11, 2019-06-14:
Tried running the code on [mint](https://ubuntu-mate.org/download/) both aarch32 ARMv7 and `aarch64 ARMv8` and they worked.
So the architecture of the OS is not the problem.

guyluz11, 2019-06-15:
Found out that [WiringNP](https://github.com/friendlyarm/WiringNP) and [WiringPi](https://github.com/WiringPi/WiringPi) are not the same.
On the WiringNP git hub page
> This is a GPIO access library for NanoPI. It is based on the WiringOP for Orange PI which is based on original WiringPi for Raspberry Pi. 

So WiringNP is based on WiringPi, so in theory it will be much easier to add support for it.
Can you add support or point me in the direction to build it myself with WiringNP insted ?

danrubel, 2019-08-11:
Thank you for all the useful information. This is not something I have investigated yet. I have not had time to focus on this package recently, but hope to do so sometime in the fall.

danrubel, 2024-10-07:
Version 0.10 now relies on `libgpiod` so that it will work on RPi v5.
Please let me know if it now works in your situation.