danrubel/rpi_gpio.dart #17: flutter pub global run rpi_gpio:build_lib failing. "The system cannot find the file specified."

ukjamesh, 2020-12-31:
Hi, I notice this project hasn't had much traction for a while, however any ideas what would be causing this?

`C:\>flutter pub global run rpi_gpio:build_lib --verbose
[  +93 ms] executing: [C:\flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[  +57 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +1 ms] 78910062997c3a836feee883712c241a5fd22983
[        ] executing: [C:\flutter/] git tag --points-at HEAD
[  +43 ms] Exit code 0 from: git tag --points-at HEAD
[   +1 ms] 1.22.5
[   +6 ms] executing: [C:\flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[  +39 ms] Exit code 0 from: git rev-parse --abbrev-ref --symbolic @{u}
[   +1 ms] origin/stable
[   +1 ms] executing: [C:\flutter/] git ls-remote --get-url origin
[  +34 ms] Exit code 0 from: git ls-remote --get-url origin
[        ] https://github.com/flutter/flutter.git
[  +58 ms] executing: [C:\flutter/] git rev-parse --abbrev-ref HEAD
[  +36 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] stable
[  +49 ms] Artifact Instance of 'AndroidMavenArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +2 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +12 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[        ] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[        ] Artifact Instance of 'AndroidMavenArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[        ] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
[  +18 ms] Using C:\flutter\.pub-cache for the pub cache.
[        ] executing: C:\flutter\bin\cache\dart-sdk\bin\pub.bat global run rpi_gpio:build_lib --verbose
Dart VM... C:\flutter\bin\cache\dart-sdk\bin\dart
Dart SDK... C:\flutter\bin\cache\dart-sdk
**CreateProcessW failed 2
Unhandled exception:
ProcessException: The system cannot find the file specified.

  Command: C:\flutter\bin\cache\dart-sdk\bin\pub list-package-dirs
#0      _ProcessImpl._runAndWait (dart:io-patch/process_patch.dart:473:7)
#1      _runNonInteractiveProcessSync (dart:io-patch/process_patch.dart:616:18)
#2      Process.runSync (dart:io-patch/process_patch.dart:66:12)
#3      main (file:///C:/flutter/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.5.0/bin/build_lib.dart:30:27)
#4      _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:299:32)
#5      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)**
[ +317 ms] "flutter global" took 375ms.
[   +6 ms] pub finished with exit code 255
[   +2 ms]
           #0      throwToolExit (package:flutter_tools/src/base/common.dart:14:3)
           #1      _DefaultPub.interactively (package:flutter_tools/src/dart/pub.dart:375:7)
           <asynchronous suspension>
           #2      PackagesForwardCommand.runCommand (package:flutter_tools/src/commands/packages.dart:260:15)
           #3      FlutterCommand.verifyThenRunCommand (package:flutter_tools/src/runner/flutter_command.dart:977:18)
           #4      _rootRunUnary (dart:async/zone.dart:1198:47)
           #5      _CustomZone.runUnary (dart:async/zone.dart:1100:19)
           #6      _FutureListener.handleValue (dart:async/future_impl.dart:143:18)
           #7      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:696:45)
           #8      Future._propagateToListeners (dart:async/future_impl.dart:725:32)
           #9      Future._completeWithValue (dart:async/future_impl.dart:529:5)
           #10     Future._asyncCompleteWithValue.<anonymous closure> (dart:async/future_impl.dart:567:7)
           #11     _rootRun (dart:async/zone.dart:1190:13)
           #12     _CustomZone.run (dart:async/zone.dart:1093:19)
           #13     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
           #14     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
           #15     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
           #16     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
           #17     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118:13)
           #18     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:169:5)`

I've tried clone flutter again to a new location and setting a new path variable but it made no difference.

Thanks

danrubel, 2020-12-31:
It *should* be able to automatically find the cached pub directory and build there, but it seems that step is broken.
I'll work on getting that fixed.

In the meantime, try this and let me know if it works around the problem.
```
cd $HOME/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.5.0
dart pub global run rpi_gpio:build_lib
```

ukjamesh, 2020-12-31:
No same problem. 

```
PS C:\flutter\.pub-cache\hosted\pub.dartlang.org\rpi_gpio-0.5.0> dart pub global run rpi_gpio:build_lib --verbose
Dart VM... C:\flutter\bin\cache\dart-sdk\bin\dart.exe
Dart SDK... C:\flutter\bin\cache\dart-sdk
CreateProcessW failed 2
Unhandled exception:
ProcessException: The system cannot find the file specified.

  Command: C:\flutter\bin\cache\dart-sdk\bin\pub list-package-dirs
#0      _ProcessImpl._runAndWait (dart:io-patch/process_patch.dart:473:7)
#1      _runNonInteractiveProcessSync (dart:io-patch/process_patch.dart:616:18)
#2      Process.runSync (dart:io-patch/process_patch.dart:66:12)
#3      main (file:///C:/Users/james/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/rpi_gpio-0.5.0/bin/build_lib.dart:30:27)
#4      _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:299:32)
#5      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)
PS C:\flutter\.pub-cache\hosted\pub.dartlang.org\rpi_gpio-0.5.0>
```

danrubel, 2021-01-01:
It appears that you are running this on a Windows machine?
This package is designed to run on the Raspberry Pi... not sure how well it will run on other platforms.

danrubel, 2021-01-01:
Are you running Flutter on Windows on Raspberry Pi?
If so, I may need to update the paths in `build_lib`.dart.

danrubel, 2021-07-06:
I've updated the build script to run on Windows, but I don't have a Windows machine on which to test.
I've published a new version with this change.
Please let me know if it fixes you problem.

Tealcoune, 2021-08-20:
Hi.
I have a somewhat equivalent problem with version 0.7.1 when I run the command : pub global run rpi_gpio:build_native
```
Building flutter tool...
Running pub upgrade...
Dart VM... C:\src\flutter\flutter\bin\cache\dart-sdk\bin\dart.exe
Dart SDK... C:\src\flutter\flutter\bin\cache\dart-sdk
Unhandled exception:
Unsupported operation: Illegal character in path: include\dart_api.h
#0      _Uri._checkWindowsPathReservedCharacters (dart:core/uri.dart:1644:11)
#1      _Uri._toWindowsFilePath (dart:core/uri.dart:2671:7)
#2      _Uri.toFilePath (dart:core/uri.dart:2646:38)
#3      new File.fromUri (dart:io/file.dart:248:49)
#4      main (file:///[home]/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/rpi_gpio-0.7.1/bin/build_native.dart:20:27)
#5      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:286:32)
#6      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:188:12)

```

danrubel, 2021-08-21:
@Tealcoune I've published `rpi_gpio 0.7.2` with a couple fixes to the `bin/build_native.dart` script.
Again, I don't have a Windows machine on which to test, but hopefully I've fixed the issue above.
Please let me know how it goes.

Tealcoune, 2021-08-21:
@danrubel
For "illegal character... " is fix. (I fix this problem localy with other option, I add context for join

Now I have another problem 

I modified line 24 : dartSdk.uri.resolve('bin/pub') to dartSdk.uri.resolve('bin/pub.bat') 
but
```
ProcessException: %1 is not a valid Win32 application.
  Command: C:\[HOME]\AppData\Local\Pub\Cache\hosted\pub.dartlang.org\rpi_gpio-0.7.2\lib\src\native\build_native 2 C:\flutter\bin\cache\dart-sdk
#0      _ProcessImpl._runAndWait (dart:io-patch/process_patch.dart:476:7)
#1      _runNonInteractiveProcessSync (dart:io-patch/process_patch.dart:619:18)
#2      Process.runSync (dart:io-patch/process_patch.dart:66:12)
#3      main (file:///C:/[HOME]/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/rpi_gpio-0.7.2/bin/build_native.dart:47:31)
#4      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:281:32)
#5      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:184:12)

```

danrubel, 2021-08-21:
It occurs to me now that `lib\src\native\build_native` is a bash script that runs on the Linux cmdline.
I'm guessing that it needs to be manually translated to a `build_native.bat` file that runs on the Windows cmdline
to compile the native part of `rpi_gpio` on the Raspberry Pi.

Tealcoune, 2021-08-21:
@danrubel
I created the .bat but you used specific imports from a linux environment
```
rpi_gpio_ext.cc:7:10: fatal error: sys/mman.h: No such file or directory
    7 | #include <sys/mman.h>
      |          ^~~~~~~~~~~~
compilation terminated.
```

danrubel, 2021-08-22:
@Tealcoune All the Dart code is cross platform and will run on Windows, but I should have realized that native Windows code on Raspberry Pi is nothing like native Linux code. It's not hard to rewrite the `lib/src/native/rpi_gpio_ext.cc` code to be Windows C# code that uses the Windows IoT libraries because that native shim is simple code... but I need to setup a Windows IoT environment on a Raspberry Pi to do so. If someone has the ability and bandwidth to do so and contribute it here that would be wonderful (big plea for help here), but if not then I will do so... but it will take me some time to spin up on Windows.