danrubel/rpi_gpio.dart #20: Will this work on RPi4 with Ubuntu Server 20.40.2.0 or does it require Raspbian?

wdmtaj, 2021-08-15:
Dan,
    I'm a fairly new Dart developer.  I developed an IoT app in Python for AWS, but I've decided to go 100% Dart/Flutter and port everything to GCP.  The Edge device in this system is A RPi4 running Ubuntu server 20.40.2.0 with the latest version of Dart.  I'm getting this error when I run:  pub global run rpi_gpio:build_native

Dart VM... /home/ubuntu/dart-sdk/bin/dart
Dart SDK... /home/ubuntu/dart-sdk
Building library in /home/ubuntu/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.7.0/lib
rpi_gpio version 0.7.0
Not running on Raspberry Pi... skipping build
Unhandled exception:
Aborting build
#0      assertRunningOnRaspberryPi (file:///home/ubuntu/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.7.0/bin/build_native.dart:99:3)
#1      main (file:///home/ubuntu/.pub-cache/hosted/pub.dartlang.org/rpi_gpio-0.7.0/bin/build_native.dart:42:3)
#2      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:281:32)
#3      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:184:12)

Thank you!
Bill Miller
1-678-521-5073

danrubel, 2021-08-19:
I've published a new `rpi_gpio 0.7.1` that has a more lenient build script.
Please upgrade to `0.7.1` and let me know.

danrubel, 2022-11-14:
This package has been updated to use `dart:ffi` and now runs on Dart SDK 2.18