danrubel/rpi_gpio.dart #21: Use `dart:ffi` for native code?

simolus3, 2021-11-13:
Thanks for your awesome work on this package.

This package uses `dart-ext` imports to access native code, a mechanism that is deprecated and will be removed in Dart 2.15. I wonder if you have considered using `dart:ffi` to bind to native code instead?
I'm happy to send in a PR if you're interested.

danrubel, 2021-11-14:
I have not had time to look into `dart:ffi`, but I definitely want to move in that direction.
It would be wonderful if you have the time to pull together a PR.
No worries if you are busy... I will get to it when I can.

simolus3, 2021-11-21:
I broke my PI out of stupidity (short-circuited the wrong pins) so it looks like I won't be finishing a PR for this soon :( Sorry for over-promising!

danrubel, 2021-11-22:
No worries!
I'll get to it when I can.

closed 2022-11-14 by danrubel