danrubel/rpi_gpio.dart #24: Some input pins not being initialized

khanhnwin, 2023-03-13:
Hey Dan,

I'm using rpi_gpio in conjunction with the [Waveshare Game HAT](https://www.waveshare.com/game-hat.htm) on a Raspberry Pi 4B. I'm puzzled why my code is successfully reading input from some of the pins, but not others. I printed out the initial stream values below [3] and noticed that the working pins have an initial value of true, but the non-working pins have an initial value of false. In the code [2], I use Pull.up to initialize the input so I expected all of the pins to be initialized to a true state. 

I suspected maybe my pins were defective, so I tried running a python script written with the RPi.GPIO library and that worked. Beyond that, if I run the Python script _and then_ my Dart script, my Dart code works and the pins are initialized correctly and I get input from all the buttons. 

Have you encountered this issue before? Do you have any idea what I might be overlooking here?

Thanks for your help! 

Pins & State
- 7, working
- 12, not working
- 16, not working
- 29, working
- 31, working
- 33, working
- 35, working
- 32, not working
- 36, not working
- 37, not working
- 38, not working
- 40, not working

[1] GPIO Enum
```
enum GameHatGPIO {
  SELECT_BUTTON(7, GameControlEvent.SELECT),
  TL_BUTTON(12, GameControlEvent.LEFT_SHOULDER),
  TR_BUTTON(16, GameControlEvent.RIGHT_SHOULDER),
  DPAD_UP_BUTTON(29, GameControlEvent.UP),
  DPAD_DOWN_BUTTON(31, GameControlEvent.DOWN),
  DPAD_LEFT_BUTTON(33, GameControlEvent.LEFT),
  DPAD_RIGHT_BUTTON(35, GameControlEvent.RIGHT),
  B_BUTTON(32, GameControlEvent.B),
  X_BUTTON(36, GameControlEvent.X),
  A_BUTTON(37, GameControlEvent.A),
  Y_BUTTON(38, GameControlEvent.Y),
  START_BUTTON(40, GameControlEvent.START);

  final int pin;
  final GameControlEvent event;
  const GameHatGPIO(this.pin, this.event);
}
```

[2] Initializing code
```
Future initialize() async {
    final gpio = await initialize_RpiGpio();

    GpioInput buttonController;

    for (GameHatGPIO button in GameHatGPIO.values) {
      buttonController = gpio.input(button.pin, Pull.up);
      buttonController.values.listen((bool buttonValue) {
        if (buttonValue == false) {
          print('${button.event} was pressed');
          // button was pressed
          events.add(GameInput(button.event, EventState.PRESSED));
        } else if (buttonValue == true) {
          print('${button.event} was released');
          // button was released
          events.add(GameInput(button.event, EventState.RELEASED));
        }
      });
    }
  }
```

[3] Initial stream values output:
```
flutter: GameControlEvent.SELECT was released (TRUE)
flutter: GameControlEvent.LEFT_SHOULDER was pressed (FALSE)
flutter: GameControlEvent.RIGHT_SHOULDER was pressed (FALSE)
flutter: GameControlEvent.UP was released (TRUE)
flutter: GameControlEvent.DOWN was released (TRUE)
flutter: GameControlEvent.LEFT was released (TRUE)
flutter: GameControlEvent.RIGHT was released (TRUE)
flutter: GameControlEvent.B was pressed (FALSE)
flutter: GameControlEvent.X was pressed (FALSE)
flutter: GameControlEvent.A was pressed (FALSE)
flutter: GameControlEvent.Y was pressed (FALSE)
flutter: GameControlEvent.START was pressed (FALSE)

```

[4] python script:
```
import RPi.GPIO as GPIO # Import Raspberry Pi GPIO library

def button_callback(channel):
    print("Button was pushed!")

PINS = [7, 12, 16, 29, 31, 33, 35, 32, 36, 37, 38, 40]

GPIO.setwarnings(False) # Ignore warning for now
GPIO.setmode(GPIO.BOARD) # Use physical pin numbering

for pin in PINS:
    GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_UP) # Set pin 10 to be an input pin and set initial value to be pulled low (off)

GPIO.cleanup() # Clean up
```

danrubel, 2023-03-22:
I need to compare the Python code that [initializes the pin's pull up/down](https://github.com/sarnold/RPi.GPIO/commit/707f0e59dca8422a82e13af143243b3447a1e013) 
with the code in this package [that does the same thing](https://github.com/danrubel/rpi_gpio.dart/blob/main/lib/src/native/rpi_gpio_ext.cc#L266-L279) to see if I can spot any problem.

Not sure when I'll be able to get to this, but something to investigate.

khanhnwin, 2023-03-27:
Thanks Dan! Yeah, unfortunately I had trouble finding those specs as well. I'll try taking a look around the Dart code and report back if I spot anything that stands out.

Appreciate your help!

khanhnwin, 2023-05-15:
Hey Dan,

I did some digging, perhaps these resources will be useful to you. It seems like the controls GPIO pull up/down/enable have been changed on the Pi 4s. That explains why the Python code that [initializes the pin's pull up/down](https://github.com/sarnold/RPi.GPIO/commit/707f0e59dca8422a82e13af143243b3447a1e013) which you linked to checks whether it's running on a a Pi 4 or not. That might be it, but even then, it's not clear to me why my current Dart code initializes some pins and not others. I would have expected none of the pins to be initialized. This GPIO code is a little outside of my wheelhouse, would love your input!

Thanks!

Khanh

Here's a few more resources that I dug found:
- [RPi.GPIO Ticket](https://sourceforge.net/p/raspberry-gpio-python/tickets/168/)
- [raspi-gpio](https://github.com/RPi-Distro/raspi-gpio/commit/80fa7d04eafb3ea34fc6f2d32de5f1873b5fb369)

danrubel, 2023-05-16:
This is very helpful. Thank you!

I can't get to this for a couple weeks, but the information you provided should enable me to add support for the Pi 4. I'll keep you posted as I can make progress.

khanhnwin, 2023-07-26:
Hi @danrubel! Just wanted to check in and see if you've been able to take a look at this? If there's anything else I can do to assist, please let me know! Excited to land a sample for the Raspberry Pi GameHAT :) https://www.youtube.com/watch?v=0CCVB31feO0&t=117s

danrubel, 2023-08-26:
Wow! Thanks for the link. Very cool and nice work!

I have plenty of older RPi models, but finally purchased a RPi 4 so that I can test and fix this bug.
Now I need to allocate time to setup the new pi, install everything, and fix.
Will keep you posted here.

danrubel, 2023-10-22:
I've fixed the gpio input handling on Raspberry Pi 4B and published version 0.9.1 with this fix.

Thank you again for sharing the native code used by the equivalent python module.
I heavily leaned on that code to fix gpio input handling.

closed 2023-10-22 by danrubel

khanhnwin, 2023-10-26:
Thank you, Dan! Thanks so much for your help on this! I really appreciate it. 

Once the sample code published, you'll be the first to know! :)