danrubel/rpi_gpio.dart #25: Set PWM frequency

Levi-Lesches, 2023-07-27:
As of  `v0.8.1`, the `GPIOPwm` class has a `dutyCycle` setter, but I can't find anything to set the frequency. Are there plans to add such a feature? 

I checked the native code: 
https://github.com/danrubel/rpi_gpio.dart/blob/42cc561b28088434b6abfc130ec2ec3cdbf68b51/lib/src/native/rpi_gpio_ext.cc#L300

Seems like this is hard-coded

danrubel, 2023-08-17:
Good idea. Yes, this seems like something I can add in the fall.
Code contributions are most welcome as well.

Viserion-stack, 2024-04-12:
is there any update on that?

danrubel, 2024-04-15:
No, nothing yet.
I'm hoping to publish 1.0 soon with support for Dart 3.x,
but have not had time to work on the PWM feature at all.

Viserion-stack, 2024-04-16:
[rpi_gpio.dart/lib/src/native/rpi_gpio_ext.cc](https://github.com/danrubel/rpi_gpio.dart/blob/42cc561b28088434b6abfc130ec2ec3cdbf68b51/lib/src/native/rpi_gpio_ext.cc#L300)
Line 300 in [42cc561](https://github.com/danrubel/rpi_gpio.dart/commit/42cc561b28088434b6abfc130ec2ec3cdbf68b51)
 pwmSetClock(32);                                     // 19.2 / 32 = 600KHz - Also starts the PWM

Tried to change clock to other value but seems it now working. Basically i just want to set freq to ~50Hz but seems it's more complex than I thought. Any idea how I can reach this?

lengyuekongliu, 2024-08-28:
Is there any new information?